Herring-UGAECSE-2920/Class-Instructions #3: What is the Linear Regulator for?

mlolson8, 2021-01-27:


jealford, 2021-01-27:
@mlolson8 This is a great question, we'll actually discuss this in class on Friday.

> are we limiting the Output Driver voltage to 5V?

Short answer, yes. 

Vcc2 is the power supply for the motor and it should be given the rated voltage of the motor. When first testing this H-bridge we assumed (the specs for this commercial plotter are sparse) the z-axis stepper was 12V and so supplied 12V to pin8. This caused the motor to hum and we eventually fried the driver. We've inferred that the stepper runs on 5V because supplying 5V caused the motor to functional properly, hence the regulator.

mlolson8, 2021-01-27:
Awesome thanks for the info!

closed 2021-01-27 by mlolson8