Herring-UGAECSE-2920/Class-Instructions #4: Import Issues

s-muskan, 2021-01-27:
I am unable to compile rgb1602. I keep getting an error for the import statement wiringpi. I have already tried to install wiring pi, however, it is still not compiling for me. I was wondering what I could be doing wrong.

ktkoffroth, 2021-01-27:
@s-muskan Hey! can you quickly make sure your in the `plotter` Python Virtual Environment we had you guys set up in deliverable 2? 

In a terminal, it should say (plotter) on the right side of the prompt. You can enable the virtual environment by executing:

```
workon plotter
```

s-muskan, 2021-01-27:
I am. However, it doesn't let me install Rpi.GPIO and wiringPi. A part of the error I am getting is as follows:

  fatal error: too many errors emitted, stopping now [-ferror-limit=]
  20 errors generated.
  error: command 'clang' failed with exit status 1
  ----------------------------------------
  ERROR: Failed building wheel for RPI.GPIO
  Running setup.py clean for RPI.GPIO
Failed to build RPI.GPIO

ktkoffroth, 2021-01-28:
can you post the entire error if possible? 

Also, make sure your internet connection is good. You can check for a connection by running the command `ping google.com` which should return a ping-time to google servers if you have a connection.

Also, RPI.GPIO and wiringpi should have already been installed via the `pip3` command as part of the Python Environment and Dependencies section [in deliverable 2 here](https://github.com/Herring-UGAECSE-2920/Class-Instructions/blob/main/deliverables/setup/pi_setup.md).

s-muskan, 2021-01-28:
Hi, yes I can. Its really long, but the entire error is this:

(plotter) Muskans-MacBook-Air:Group-8 muskan$ pip3 install wiringpi
Collecting wiringpi
  Using cached wiringpi-2.60.0.tar.gz (130 kB)
Building wheels for collected packages: wiringpi
  Building wheel for wiringpi (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /Users/muskan/.virtualenvs/plotter/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/5b/t5x60g617bd_yjn2cbzlpqmr0000gn/T/pip-install-et6o4wcv/wiringpi_d359aebe2fb94fd78f7601724c45604c/setup.py'"'"'; __file__='"'"'/private/var/folders/5b/t5x60g617bd_yjn2cbzlpqmr0000gn/T/pip-install-et6o4wcv/wiringpi_d359aebe2fb94fd78f7601724c45604c/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /private/var/folders/5b/t5x60g617bd_yjn2cbzlpqmr0000gn/T/pip-wheel-sq8h9pft
       cwd: /private/var/folders/5b/t5x60g617bd_yjn2cbzlpqmr0000gn/T/pip-install-et6o4wcv/wiringpi_d359aebe2fb94fd78f7601724c45604c/
  Complete output (104 lines):
  running bdist_wheel
  running build
  running build_py
  running build_ext
  building '_wiringpi' extension
  creating build
  creating build/temp.macosx-10.14.6-x86_64-3.8
  creating build/temp.macosx-10.14.6-x86_64-3.8/WiringPi
  creating build/temp.macosx-10.14.6-x86_64-3.8/WiringPi/devLib
  creating build/temp.macosx-10.14.6-x86_64-3.8/WiringPi/wiringPi
  clang -Wno-unused-result -Wsign-compare -Wunreachable-code -fno-common -dynamic -DNDEBUG -g -fwrapv -O3 -Wall -iwithsysroot/System/Library/Frameworks/System.framework/PrivateHeaders -iwithsysroot/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/Headers -arch arm64 -arch x86_64 -IWiringPi/wiringPi -IWiringPi/devLib -I/Users/muskan/.virtualenvs/plotter/include -I/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.8/include/python3.8 -c WiringPi/devLib/piFace.c -o build/temp.macosx-10.14.6-x86_64-3.8/WiringPi/devLib/piFace.o
  In file included from WiringPi/devLib/piFace.c:28:
  In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/include/stdio.h:64:
  In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/include/_stdio.h:68:
  /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/include/sys/cdefs.h:807:2: error: Unsupported architecture
  #error Unsupported architecture
   ^
  In file included from WiringPi/devLib/piFace.c:28:
  In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/include/stdio.h:64:
  In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/include/_stdio.h:71:
  In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/include/_types.h:27:
  In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/include/sys/_types.h:33:
  /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/include/machine/_types.h:34:2: error: architecture not supported
  #error architecture not supported
   ^
  In file included from WiringPi/devLib/piFace.c:28:
  In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/include/stdio.h:64:
  In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/include/_stdio.h:71:
  In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/include/_types.h:27:
  /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/include/sys/_types.h:55:9: error: unknown type name '__int64_t'
  typedef __int64_t       __darwin_blkcnt_t;      /* total blocks */
          ^
  /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/include/sys/_types.h:56:9: error: unknown type name '__int32_t'; did you mean '__int128_t'?
  typedef __int32_t       __darwin_blksize_t;     /* preferred block size */
          ^
  note: '__int128_t' declared here
  /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/include/sys/_types.h:57:9: error: unknown type name '__int32_t'; did you mean '__int128_t'?
  typedef __int32_t       __darwin_dev_t;         /* dev_t */
          ^
  note: '__int128_t' declared here
  /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/include/sys/_types.h:60:9: error: unknown type name '__uint32_t'; did you mean '__uint128_t'?
  typedef __uint32_t      __darwin_gid_t;         /* [???] process and group IDs */
          ^
  note: '__uint128_t' declared here
  /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/include/sys/_types.h:61:9: error: unknown type name '__uint32_t'; did you mean '__uint128_t'?
  typedef __uint32_t      __darwin_id_t;          /* [XSI] pid_t, uid_t, or gid_t*/
          ^
  note: '__uint128_t' declared here
  /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/include/sys/_types.h:62:9: error: unknown type name '__uint64_t'
  typedef __uint64_t      __darwin_ino64_t;       /* [???] Used for 64 bit inodes */
          ^
  /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/include/sys/_types.h:68:9: error: unknown type name '__darwin_natural_t'
  typedef __darwin_natural_t __darwin_mach_port_name_t; /* Used by mach */
          ^
[... cut ...] wiringpi_d359aebe2fb94fd78f7601724c45604c/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /private/var/folders/5b/t5x60g617bd_yjn2cbzlpqmr0000gn/T/pip-record-c8orgwkn/install-record.txt --single-version-externally-managed --compile --install-headers /Users/muskan/.virtualenvs/plotter/include/site/python3.8/wiringpi Check the logs for full command output.

ktkoffroth, 2021-01-28:
Do you happen to be attempting to install this on your Macbook? These errors are very MacOS specific, and the wiringpi package is made for the Raspberry Pi.

s-muskan, 2021-01-28:
Yes, I am working in Visual Studio Code on a Macbook. Is the goal to code through the raspberry pi IDE using the nano text edit? I was hoping to be able to compile the file on my end before pushing it on to github so my group member can try to run it on the raspberry pi. We tried using the nano text edit and ran into a different error dealing with attributes.

ktkoffroth, 2021-01-28:
Well first, Python is not a compiled language, you can edit the script and run it immediately. However, since most of your code will be working with hardware, you'll need to run it on the Pi to test/debug. This means if you would like to make changes to the code remotely, you would upload to GitHub and whoever has access to the Pi would `git pull` your repository on the Pi with the new changes. Using something like SSHFS would allow you to edit code on the Pi with an editor on your computer. You can read more about this flow in the [working on the pi doc](https://github.com/Herring-UGAECSE-2920/Class-Instructions/blob/main/deliverables/setup/working_on_pi.md).

As far as your error goes, make sure you're in the Python virtual environment we setup in the [pi setup doc](https://github.com/Herring-UGAECSE-2920/Class-Instructions/blob/main/deliverables/setup/pi_setup.md) by using the command `workon plotter`.

If that doesn't resolve it, make sure you have wiringpi and any other of the Python dependencies installed as listed in the Pi Setup documentation linked above.

s-muskan, 2021-01-29:
Okay, thank you!

closed 2021-01-29 by s-muskan