Herring-UGAECSE-2920/Class-Instructions #5: sudo apt-get install vim-gtk Command Issue

tristanthomas3, 2021-01-28:
We put in this command from the guide for the LCD and we are seeing a bunch of fetching errors. It mentions at the end to try apt-get update and this doesn't work either. We're connected to Paws-secure but bridged through a computer through VNC. Could something with the connection be causing an issue or something else?

ktkoffroth, 2021-01-28:
@tristanthomas3 As far as Vim goes, this is just the editor the author of the guide recommends when using the Pi. It's a command-line editor/ide, which if you're using VNC, wouldn't be as useful as the Thonny IDE we set up in the Pi_Setup guide. In short, you can skip the step.

As far as your internet connection goes, we do not recommend attempting to connect the Pi to PAWS-Secure, as it's spotty at best and has problems with p2p connections such as VNC and SSH. If you have your laptop connected to the internet, the Ethernet bridge should be passing that connect through to your Pi.

ktkoffroth, 2021-01-28:
@tristanthomas3 Has this problem been resolved?

closed 2021-01-29 by ktkoffroth