Herring-UGAECSE-2920/Class-Instructions #9: Cannot import stepper

jss83507, 2021-02-03:
For deliverable 4, we are trying to create code to control the motors, but are getting an error "cannot import name 'stepper' from...". We have run the default code that does not require stepper successfully, but the error persists whenever we try to import the stepper object.

jealford, 2021-02-03:
Can you post the code that is throwing the error?

jss83507, 2021-02-03:
> Can you post the code that is throwing the error?

This is the beginning of our code where the importing is done:
import time
import board
from adafruit_motorkit import stepper
from adafruit_motorkit import MotorKit

jealford, 2021-02-03:
Double check your imports. 
https://learn.adafruit.com/adafruit-dc-and-stepper-motor-hat-for-raspberry-pi/using-stepper-motors

jss83507, 2021-02-03:
We have been using that page to diagnose our import problems, and have used the same statements as in the page. It works with just the forward direction (code that does not require stepper), but when we import stepper using the same import statement as the documentation we get the error.

ktkoffroth, 2021-02-03:
on the [Using Stepper Motors](https://learn.adafruit.com/adafruit-dc-and-stepper-motor-hat-for-raspberry-pi/using-stepper-motors) page it is mentioned that the correct import is:

```
from adafruit_motor import stepper
```
just drop the `kit` and it should work.

If it doesn't, it means something didn't install, in that case use this command to install the `adafruit_motor` library:

```
pip3 install adafruit-circuitpython-motor
```

jss83507, 2021-02-03:
We've found a way to implement the stepper motor movement in both directions without the use of the "Stepper" function. We will double check our motor library install as well and keep trying to implement the code as outlined in the documentation. Thank you!

closed 2021-02-03 by jss83507