Herring-UGAECSE-2920/Class-Instructions #10: 1.5 V drop after coming out of regulator

UaNotJosh, 2021-02-09:
When using the regulator by itself, it outputs the 5 V that we want it to. However, when we plug it into the H-bridge for the logic and enable pins, the voltage suddenly drops to 3.5 V. We are able to get sound out of the motor using gpio and feel like something inside is rotating, but it isn't rotating to raise or lower the pen at all. According to the documentation, it needs at least 4.5 V for the logic to function, so we feel that it is an issue coming from the 3.5 V but we don't understand why it is dropping. There is a chance that we aren't supposed to use the same 5 V for the logic and the enable pins, but we do not know what to use otherwise.

ktkoffroth, 2021-02-09:
Hey @UaNotJosh and @Herring-UGAECSE-2920/group-3, can you send me a schematic/picture of your setup? The linear regulator or H-Bridge chip may not be setup correctly on your end, so I want to eliminate that.

The VCC1 and both EN pins can be powered from the Pi or any other 5V source. VCC2 **MUST** be connected with the 5V linear regulator, as that is the VCC/Power for the stepper motor, which can draw far more than the Pi can supply.

tristanthomas3, 2021-02-10:
Hey Kevin, 

Thanks for the response. So I will post what I talked to you about in case it will help other people. So we know now that the Vcc2 that we were using (12V) is not correct and we need to change it to 5. Do you think that was what was dropping the voltage of our regulator? Perhaps we were loading the 12V input too much causing it to drop the output voltage of the regulator? I can still post the diagram if needed, but it sounds like we need to change that Vcc2 to know if there is actually another issue or if that was the only thing.

ktkoffroth, 2021-02-10:
Yes, go ahead and try that out, and post here again if the issue persists.

closed 2021-02-10 by ktkoffroth