Herring-UGAECSE-2920/Class-Instructions #11: Follow Up

tristanthomas3, 2021-02-10:
So regarding our last issue, I changed the voltage on pin 8 from 12V to 5. We are still seeing our regulator voltage drop when the output is connected to the H-bridge chip. 
Here's the wiring diagram we have right now:
[image]

Please note, in our diagram, we have not figure out how to change the pin numbers to the right physical location. So our pin 1 is top left, pin 2 top right, pin 15 bottom left, pin 16 bottom right. I know this is annoying to compare for you; I am trying to work on making it in a different program. However, we are making sure to connect pin 7 for example on the wiring diagram to the actual pin 7 of the chip which is the 2nd from the bottom on the left, so we do not think that is the issue.

Similarly, here is our setup:

[image]
I know this is hard to see but I will summarize the connections we have:
Pin1: 5V from output of regulator
Pin 2,3,6,7,10,11,14,15: These 8 pins are not connected here but they all either go to the GPIO of the pi or to the motor
Pin 4,5,12,13: Connected to ground (we are using the same ground as the 12V wall supply)
Pin 8: Now connected to 5V output of regulator
Pin 9: 5V output from regulator 
Pin 16: 5V output from regulator 

Note also, the voltage is dropping as shown in the diagram, meaning even if the pi and motor are not connected the regulator voltage still drops when connected to the h bridge. 

We have thought that maybe either the enable pins or the logic supply could be provided 5V from the pi, but would that really make any difference? We are using the 5V output from the regulator for both enable pins, the logic pin (Vcc1), and the Vcc2 pin. From the last issue, it seems this shouldn't be a problem. Please let me know if you need anymore detail about our issue.

Thanks in advance!

ktkoffroth, 2021-02-10:
So, your circuit looks correct, and you say the voltage drops whenever you hook up the H-Bridge. Did you guys attempt to drive the motor with 12V to VCC2? If you did, it could have killed the H-Bridge chip.

Have you guys tried driving the motor with your current circuit?

tristanthomas3, 2021-02-11:
Yes we did have vcc2 setup with 12V yesterday and we tried to drive the motor and we heard noise but it wasn't actually spinning. 

We could try driving the motor now with the 5V on vcc2 but the regulator is no longer providing 5V when connected to the h bridge at all. Whether we give 5V or 12V to vcc2, the output pin of the regulator does not output 5V truly to the h bridge anymore. In that case, I think it's safe to say it's probably not going to be able to drive the motor because it's in the same state as yesterday. Perhaps we could just go ahead and try with a new h bridge?

ktkoffroth, 2021-02-11:
Can you please try it? I'm away from my plotter at the moment, so I can't confirm the correct voltage level the H-Bridge pulls when not driving a motor. If it still does not work, I can meet you all tomorrow to try a new regulator and H-Bridge chip. Also, please post the script you're using to drive the motor.

UaNotJosh, 2021-02-11:
Hi Kevin, our code for the z-axis movement is in our group repository. It's under lib/MotorMovement/Z-axis-movement.py. I have two different methods that I tried in the code with one method commented out. Both methods resulted in movement within the motor as evident by the sound it was making, but none of it made the motor actually rotate.

ktkoffroth, 2021-02-11:
Hey @UaNotJosh, the code appears to be incorrect. I would do a bit more research on [how bipolar stepper motors work](https://en.wikipedia.org/wiki/Stepper_motor) and [how to use the pigpio library](abyz.me.uk/rpi/pigpio/python.html).

UaNotJosh, 2021-02-11:
Okay, thank you we are developing another idea for this now. However, going back to the regulator, could you explain further about its operation? Our understanding is that if the regulator is given a 12V input, then a 5V output should be maintained regardless of its load. Furthermore, when the regulator is loaded with nothing but the 5V pins of the H-bridge, the voltage should not drop at all right? Is our logic wrong here?

Can we still plan to meet tomorrow for another regulator if possible? Tristan talked to Dr. Herring earlier today and was told if the regulator was only connected to the 5V pins of the H-bridge, the voltage should not drop that low.

tristanthomas3, 2021-02-11:
Another thought Kevin... do we need to connect the ground of the pi to the ground we are using for the regulator?

ktkoffroth, 2021-02-11:
@tristanthomas3 That would be a good idea yes.

@UaNotJosh [Here](https://www.ti.com/general/docs/suppproductinfo.tsp?distId=26&gotoUrl=http%253A%252F%252Fwww.ti.com%252Flit%252Fgpn%252Flm340) is a download for the datasheet for your linear regulator. As I said before, I'm away from my plotter, so I cannot confirm if the voltage drop is normal behavior or not. I agree that it logically should not be at a lower voltage, so that would lead me to believe either your linear regulator or H-Bridge circuit is broken.

In any case, test it as soon as you can, and we can get you a new regulator and H-Bridge should it still be an issue.

tristanthomas3, 2021-02-11:
So we're not going to be able to test for another few hours due to our team having classes and things like that. In the event that we do test it in the next couple hours and things still aren't working, what would be the best way to arrange a time to potentially look at using a new regulator?

Hopefully it will work and that won't be needed but we just want to be prepared

ktkoffroth, 2021-02-11:
Yes, I'll be on campus till late, just message me on GroupMe

UaNotJosh, 2021-02-11:
Hi Kevin, we're still having issues even with a new regulator, h-bridge, and code. For some reason, even when we disconnect the regulator altogether, we still se a voltage of about 3V on pin 8 of the h-bridge even though nothing is plugged directly into that pin. We are using 5V from the pi to power everything that needs 5V but VCC2. Also, for some reason, when we connect the ground of the pi to the ground from the power supply, we see an initial output voltage of 5V from the regulator but it quickly drops and gets really hot. We don't think we fried it however, because it still works when using it alone. I posted the new code in the same place as the other one and got rid of the other one to not make it confusing.

ktkoffroth, 2021-02-11:
Alright, since the new voltage regulator and H-bridge chip did not fix your voltage issue, I'm certain it can be ignored. If your hardware setup is accurate to your circuit diagram, you have nothing to worry about.

Your code does need work. You seem to have the right idea about how to control it w/ GPIO, but you need to look through the Python pigpio library reference in the [usage section](http://abyz.me.uk/rpi/pigpio/python.html) again.  Also take a look at the [set_mode section](abyz.me.uk/rpi/pigpio/python.html#set_mode). In addition to all that, make sure the pigpio daemon is running on your pi. You should also include a time delay after setting the pin modes (high/low) for each step; your program is changing the pin mode before pigpio can even output.

If you continue to have problems with the library, you should look into the [RPi,GPIO](https://sourceforge.net/p/raspberry-gpio-python/wiki/Outputs/) library.

tristanthomas3, 2021-02-11:
Okay, we will make those changes. The only thing we are concerned about is that we are measuring a voltage off the H-bridge with nothing powered. This seems to be messing with the regulator.

As an idea, could we take our updated code, with the changes you listed above, and try to use a 5V benchtop power supply first and see if it works, because it should supply plenty of power right?

ktkoffroth, 2021-02-11:
Connect all your grounds. Connect the EN and VCC1 pins to the 5V rail on the Pi. Connect VCC2 to the linear regulator output. With your new linear regulator and H-Bridge chip, there should be no hardware issues as long as everything is connected properly. Push forward w/ working on the code for now. You can fix any hardware issues after that is resolved.

UaNotJosh, 2021-02-11:
We got it Kevin. Thank you so much for all of your help!

closed 2021-02-12 by ktkoffroth