Herring-UGAECSE-2920/Class-Instructions #12: Z-axis motor is not turning

tbr60702, 2021-02-10:
Here is how we currently have the H-Bridge connected to GPIO pins, linear regulator, and Z-axis motor except for Vcc2. Vcc2 (pin 8) is attached to the linear regulator output instead of 12V. 
[image]

Our problem is that our z-axis motor is not spinning when we want it to. It applies tension to the pulley in the downward direction like it is trying to, but it does not actually turn.

With the Z-axis motor disconnected, the H-Bridge outputs ~5V when we want it to based of off our script. With the Z-axis motor connected, the H-Bridge outputs ~3.6V, ~360mA when the output is high. 

Are we having a wiring issue or is this something in our script that we need to change?

ktkoffroth, 2021-02-10:
I see you have your Enable pins attached to GPIO on the Pi. Are these pins set to 5V in your script?

ktkoffroth, 2021-02-11:
New information: you will not be able to use the GPIO to drive the EN pins on the H-Bridge chip. [As detailed here] the raspberry pi GPIO has an output voltage of 3.3V, and the EN pins require 5V. Please connect your EN pins to one of the 5V pins on the Pi or to the linear regulator.

tbr60702, 2021-02-11:
Here is our script:
        import RPi.GPIO as GPIO
        from time import sleep
        
        GPIO.setwarnings(False)
        GPIO.setmode(GPIO.BCM)
        GPIO.setup(27, GPIO.OUT)
        GPIO.setup(25, GPIO.OUT)
        GPIO.setup(4, GPIO.OUT)
        GPIO.setup(26, GPIO.OUT)
        GPIO.setup(5, GPIO.OUT)
        GPIO.setup(6, GPIO.OUT)
        
        EN1 = 5
        EN2 = 6
        SIG1A = 26
        SIG1B = 4
        SIG2A = 27
        SIG2B = 25
        
        
        GPIO.output(EN1, GPIO.HIGH)
        GPIO.output(EN2, GPIO.HIGH)
        while True:
            
            GPIO.output(SIG1A, GPIO.HIGH)
            GPIO.output(SIG1B, GPIO.LOW)
            GPIO.output(SIG2A, GPIO.LOW)
            GPIO.output(SIG2B, GPIO.HIGH)
            sleep(2)
            
            GPIO.output(SIG1A, GPIO.LOW)
            GPIO.output(SIG1B, GPIO.LOW)
            GPIO.output(SIG2A, GPIO.LOW)
            GPIO.output(SIG2B, GPIO.LOW)
            sleep(2)
            
            GPIO.output(SIG1A, GPIO.LOW)
            GPIO.output(SIG1B, GPIO.HIGH)
            GPIO.output(SIG2A, GPIO.HIGH)
            GPIO.output(SIG2B, GPIO.LOW)
            sleep(2)
            
            GPIO.output(SIG1A, GPIO.LOW)
            GPIO.output(SIG1B, GPIO.LOW)
            GPIO.output(SIG2A, GPIO.LOW)
            GPIO.output(SIG2B, GPIO.LOW)
            sleep(2)

Our EN pins were attached to GPIO's, so we will change our schematic and wiring for the EN pins to the 5V pins on the Pi. Our linear regulator also decreases its output from ~5V with no Z-axis motor connected to 2.5-3.5V when it is having to push the Z-axis motor. It also gets very, very hot with the Z-axis motor hooked up.

ktkoffroth, 2021-02-11:
Okay, the change for the EN pins should fix your problem. I believe the motor does not pull the full 5V unless it is being driven (I don't know for sure, as I'm away from the plotter right now), and it is normal for it to get hot like that. presumably, you won't have to move the Z-Axis motor very much in operation, so it shouldn't be an issue. You could add a small heat sink to the H-Bridge chip if you wanted to though.

tbr60702, 2021-02-11:
Ok, thank you! We will try this tomorrow night with the EN pins fed off of the PI's 5V and see what happens.

ktkoffroth, 2021-02-11:
Okay, Good Luck!

closed 2021-02-11 by ktkoffroth