Herring-UGAECSE-2920/Class-Instructions #15: Pi Wont connect to SD card

ZzzMarky, 2021-02-25:
So, a few days ago our motors were working fine with the pi but then all of sudden, the pi could not connect with the 0x61 address. The soldered connection on the motor hat was still intact and there was no smoking. However, we did notice the casing that holds our SD card was Very hot. And it seems the excessive heat on the casing caused our SD Card to be rendered useless. It was pointed out to us that this could be an issue with the poly fuses inside of the board but this issue happened a few days ago so we are thinking the poly fuses should have reset by now.

Also, We tried getting a new pi from the Makerspace but we do not have the correct power supply for this pi. It is a Raspberry pi B+ v1.2? We believe an android-like charger (for the older android phones) should do the trick but we haven't had any luck obtaining one. 

All in all, we are wondering if there is a way we could obtain a new pi and SD card Temporarily and or permanently? 
Worst-case scenario, we will continue hunting for an android like charging and try to see if we can get things working with this B+ v1.2.

ktkoffroth, 2021-02-25:
Usually a hot SD card means it's broken somehow, have you guys tried plugging it into a computer (your kit came w/ a USB SD card reader) and seeing if you can format it?

If that doesn't work, I can meet y'all sometime this afternoon to get you a new one.

ZzzMarky, 2021-02-25:
Yes, we tried the SD Card reader and nothing came up. But our SD Card is unfortunately in 2 pieces now due to the heat. So it doesn't exactly work at all anymore.

ktkoffroth, 2021-02-25:
Okay, I can meet y'all this afternoon to hand out a new card (though you guys will have to replace yours eventually, unfortunately). Can you email me w/ a time and place in Driftmeir to give it to y'all?

ZzzMarky, 2021-02-25:
Can do! Thank you! Although I believe we tried a different SD Card but it was from the Raspberry pi B+ V 1.2 we had and we did not get any response. So I think the pi's SD Card Reader may be broken. But we are definitely open to trying more SD Cards

closed 2021-02-25 by ktkoffroth