Herring-UGAECSE-2920/Class-Instructions #16: Error popping up inconsistently

UaNotJosh, 2021-02-25:
Hi Kevin and/or Jake,
For some reason, when running our etch-a-sketch code there will sometimes be an error that pops up that says that the 'Rotary' object does has no attribute '#insert callback here#'. I have #insert callback here# because the callback mentioned is not the same every time the error pops up, and even past that, the error does not always pop up. It pops up roughly 40% of the time with it sometimes running perfectly 3-4 times in a row and the error popping up 3-4 times in a row. When the error pops up, whichever function is mentioned in the error does not work, but when the error is absent, the code works perfectly. I do not understand why the error would be popping up sometimes and not every time, and am concerned that it might interfere with the overall running of the code if we have to re-run the code over and over until the error does not show.
For reference, the code in question is "Etch-a-Sketch.py" located in /Group-3/lib/Etch-a-Sketch. Please let me know if there is any more information that you need to help and thank you in advance!

ktkoffroth, 2021-02-26:
is the rotary object in question for the same encoder each time? Posting a screenshot/log of the error you're getting (I understand it's different each time, just choose one) would also be very helpful.

ktkoffroth, 2021-02-26:
Also, go ahead and define and empty long press callback for your encoders, and pass it to the pigpio_encoder library (make sure you put a line with `pass` in the function or you'll have problems).

ktkoffroth, 2021-03-01:
@UaNotJosh Let me know if you continue having this issue, me and Jake are having a similar one, so we'll let you know if we find a solution.

ktkoffroth, 2021-03-02:
@UaNotJosh So after some testing, we think the issue involves some hardware imperfections. So, if you've implemented your etch-a-sketch mode in a similar way to us, the rotary encoders don't come all the way up every time after pressed, so the `pigpio_encoder` library is trying to call one of the callbacks before your program can set it correctly (which is causing the error).

A way to fix this is by using a `try/except` block in your etch-a-sketch code, catching the error, and ignoring it until you get no errors.

closed 2021-03-05 by ktkoffroth

UaNotJosh, 2021-03-07:
@ktkoffroth where did you put your try/except block? I put mine in the sw_short functions for both encoders and I am still experiencing the error message
[image]

ktkoffroth, 2021-03-07:
Since the error happens on instantiation of the pigpio_encoder objects, you'll want to have the try/except around where you're doing that.