Herring-UGAECSE-2920/Class-Instructions #19: X motor skipping steps + I2C error message

gjs19225, 2021-03-04:
The issue that we are having is that the x stepper motor is jittering or skipping steps when in use, which causes the stepper motor to become inconsistent during calibration. Starting today, we have recevied an i2c error/message on the terminal whenever code executes for the x stepper motor, and it appears after the program terminates. This issue doesn't occur with the y stepper motor.

Here are some of the solutions that we tried:
1) We swapped the x and y motor connections on the H bridge of the motor hat, but the issue with the x stepper motor still persisted. The y stepper motor worked fine.

Here is a google drive link that contains a video and photos of our issues: https://docs.google.com/document/d/1A0w99v0MkNrjx7kmXqZUwP5rZXw5ypS1E6u_1Rq2ImY/edit?usp=sharing

Please offer us some adivce to help us overcome this issue. 

Update: the x stepper motor runs smoothly when double step is used, but issues still presist when using single step

ktkoffroth, 2021-03-04:
So, if I am understanding correctly, the X Stepper motor acts like this no matter which motor connection you use on the motor hat, and the X Motor connection works fine when pair with the Y-Stepper? 

This would indicate an issue with your X Stepper Motor to me. Can you email Dr. Herring or Dr. Trudgen about securing a replacement?

gjs19225, 2021-03-04:
To clarify the initial post, a single line of code in an infinite while loop was used to test one stepper motor at a time. It would then be modified to test for the other stepper motor after the program terminated. When the X stepper motor was being tested, it would jitter/skip steps and then an i2c error would appear on the screen after the program ended.  The i2c message appears no matter what step style is being used on the X motor. These issues still presisted when the H bridge used by the X stepper motor was swapped with the Y motor's H bridge.  For the Y stepper motor, it worked fine before and after the swap, and no error message pertaining to i2c occured. 

Now after running the same code today, it seems that the Y stepper motor is giving out the same i2c error message aswell. 

Also, a fix for the X motor jittering/skipping steps was to switch from single to double (with i2c message still appearing afterwards).  I will look into getting a replacement X stepper motor for that issue.

ktkoffroth, 2021-03-04:
I believe the I2C error is occurring because you guys aren't releasing the motors before exiting the program. I would use a try/except block in your while loop with the KeyboardInterrupt exception, and have the motors release (specified in the library) and the program shutdown when the exception is caught.

closed 2021-03-04 by gjs19225