Herring-UGAECSE-2920/Class-Instructions #20: Calibration Questions

tristanthomas3, 2021-03-06:
So for the calibration modes, our understanding is that the auto calibration essentially determines where the pen can and can not go and thus at what points the motors should stop spinning. Should this limit be at the edge of the margins of the paper or the edge of the physical gantry? Or should the motors be able to spin out to the edge of the gantry, but just not allow the pen to go down out past the margin?

For pen calibration mode, is the point that the user needs to turn to always going to be bottom right or can it be any point? If it is always bottom right, what exactly should the video show since all that is going to happen is the software will determine that point as a reference? Also, why do we need this mode, if we are tracking the position of the pen at all times using the step count, couldn't we already know where we are in relation to the true center of the paper based off that?

ktkoffroth, 2021-03-08:
The Auto-Calibration is the same as what you guys did for the etch-a-sketch, it is there to simply prevent your plotter when attempting to go where it physically cannot. 

The pen calibration mode is a safeguard for having inconsistent paper placement on the plotter. A hard-coded calibration for paper center or bottom-right would be great if you perfectly positioned the paper in the exact same spot every time, but that's not going to always be the case. Not to mention the plotter hardware we gave y'all has some variation, so the mechanical limits/paper positioning could be different for each team. The pen calibration mode ensures this problem is eliminated with the caveat of a bit of inconvenience.

tristanthomas3, 2021-03-08:
Okay, one other question, does our paper need to always have a small dot in the corner where 1 inch meets 1 inch? How else will the user know exactly where the corner of the page should be?

ktkoffroth, 2021-03-08:
That would be a great way to do it.

closed 2021-03-11 by ktkoffroth