Herring-UGAECSE-2920/Class-Instructions #22: Spontaneous motor issues

UaNotJosh, 2021-03-15:
Hi guys,
So this issue is two-fold. If I need to put them into two separate issues, please let me know and I can do so.

First, an issue that we encountered last week but still have not found a solution to. For some reason, when in our calibration mode, our motors move MUCH slower than they otherwise do. This is even worse when the motors are attempting to move at the same time as they move into stepping like individually, apart from each other rather than moving at the same time. We thought this might be an issue with the motors, but when we run the code for Etch-A-Sketch mode, they move at perfect speeds, even when moving together. We initially thought this might be an issue with the number of conditional loops that it was having to check, but it goes through the same number in our calibration mode as it does for the Etch-A-Sketch mode. This issue is not critical, as the calibration mode still operates as it should, but we feel that if it is occurring now, it may crop up later as well.

Second, we are suddenly having issues with our z-motor movement. When attempting to work on the calibration mode today, we made commands for the pen down to draw the coordinate plane in the middle, but it refuses to pen down for some reason. To try and see if this was simply a coding issue, we went back to our unedited file from a few deliverables ago where we moved the z-motor up and down for 1 second each to test. It will still not pen-down in this mode. If we move the z-motor down, it will retract upwards correctly, but the issue is only on the pen down for some reason. We have triple-checked the wiring as it has been the same for the past few weeks, with the Etch-A-Sketch continuing to work properly, but all of the sudden, pen down is dysfunctional. 

Thanks in advance for any ideas!

UaNotJosh, 2021-03-15:
Follow up on this, we think we found the second issue. After testing the GPIO outputs over and over, we found that one of the GPIO that we were using would not change how it needed to when we would write it high or low. Instead of changing between ~0V and ~3.3V, it would change from ~2.3 to ~2.5. After changing to a different GPIO, we were able to get it working. Is this just an issue of a faulty GPIO or is there something else going on here? Also, can this happen again with other GPIO's and what do we do if we run into issues with space or number of available GPIO's?

ktkoffroth, 2021-03-23:
There are several things that could cause the slowdown you're describing when I looking at your code in `Calibration.py`. 

As you mentioned, there are a ton of conditional statements in your callbacks, which while they shouldn't cause much of a slowdown, could be doing so, as string parsing and comparing is an expensive operation (and you do it like 10 times per callback).

In addition to this, I can see you have several `time.sleep()` calls in your while loop, which shouldn't be called when in your manual calibration mode, but may be anyways. Also, your  `.watch()` calls are doing nothing, as the while loop never breaks. This doesn't break the encoders though, because the `watch()` function is just a while loop itself.

The behavior you described sounds a lot like you are just doing too many operations for Python to handle in one thread (it is not a fast language, definitely not fast enough for implementing real-time applications in this way). 

I can also see you forgot to put `global xSteps` and `global ySteps` in your up/down X and Y callbacks respectively.

To remedy this, I would suggest having multiple sets of callbacks which you change at runtime, depending on the functionality needed. You may need to stop the execution of your encoders to do this as well.

closed 2021-03-25 by ktkoffroth