Herring-UGAECSE-2920/Class-Instructions #23: Clarification for Power Supply Specs

bunnyrun, 2021-03-17:
Hi, I just needed some clarification on what has been written for the power supply requirements.
This is no longer listed in Del 9 anymore, but it's still on Del 7:

> Your power supply will take in 120 VAC from the wall, and supply 12V 5A DC typical, with +25% peak output capability (should be able to output up to (12V*5A) + (12V*5A)*0.25 = 60W + 15W = 75W Maximum Output) and **+/- 5% allowable tolerance.**
_(Del. 7)_

What is the 5% allowable tolerance referring to? Is it referring to the output as in the output of 12V, 5A shouldn't vary greater than +/- 5%? Or is it referring to the component tolerances because I thought those tolerances were up to us to design:

> Note: your design must show that it works **within the tolerance ranges of your components** (key word is SHOW). For example, if you **design** a circuit with 1 resistor that has a **+/- 10% tolerance**, then your calculations must show the boundaries for min/max due to this resistor.
_(Del. 9)_

Thank you so much in advance!

ktkoffroth, 2021-03-17:
Yes, sorry about the confusion, all the 5% allowable tolerance means is that your design should work within a +/- 5% range of your ideal component values. Of course, if you chose a 10% tolerance resistor, your design would also have to work within that tolerance.

Does that make sense?

ktkoffroth, 2021-03-17:
Just went and updated deliverable 7 to make it consistent

closed 2021-03-18 by ktkoffroth