Herring-UGAECSE-2920/Class-Instructions #25: Circle Function - Motor Stepper Size

catedavis, 2021-03-26:
Our horizontal motor step size is 100 steps for 33cm. and our vertical motor step size is 100 steps for 16cm. This causes problems when trying to calculate the decision parameters for the midpoint and Bresenham circle algorithm. I have tried to slip the radius into x and y legs, but that logically does not make sense. I talked to another group and their vertical and horizontal step sizes are the same. How do we implement the circle with different x and y step sizes?

jealford, 2021-03-29:
You shouldn't have such a large difference in the x and y. I would check to make sure you're using the same step type for both.

ktkoffroth, 2021-03-29:
In addition to Jake's comment, I know that the motors are rated as 200 steps/revolution, meaning that you would need to find a correlation between this smallest step size and distance traveled. The step sizes for both motors should be the same in this regard.

closed 2021-04-01 by ktkoffroth