Herring-UGAECSE-2920/Class-Instructions #27: Trouble Importing Modules Using Launch on Startup

jss83507, 2021-04-06:
Normally, we program all of our code in Thonny, when we attempted to switch to the startup script sh file, any attempt to run the program throws errors stating that it can't find the required modules for any of our hardware dependencies. We do not get any of these modules error when running programs in Thonny, only when we attempt to run on startup. We've tried running using python, python3, py, and py-m, but have had no luck. 

Next we tried importing the modules one by one through the terminal, but we've been encountering issues when loading some of the libraries like the adafruit_motor library, the libraries may be depreciated/are no longer available from the normal download path. In that case, we tried installing adafruit-circuit(something)-motor package, and that failed to install.

Any suggestions would be GREATLY appreciated! :)

ktkoffroth, 2021-04-07:
@jss83507 Well, the Python import system is not exactly perfect, or even good. Here are a few things you can check, and some resources in case those don't work:

1. Make sure your script is working in the same Python virtual environment you guys are developing in on Thonny. This could be the `plotter` virtual environment we setup in deliverable 2, or the default one for your `pi` user. In any case, check which virtual environment you're using in Thonny.
2. If you are in the right virtual environment in your script, check the `site-packages` for that environment to make sure the correct packages are installed and available.
3. Make sure your import paths are correct. Python uses directory-based package organization, so if you're having issues importing modules that you didn't `pip`-install, make sure your script can see/access those directories in your import path.

If all of those fail to help you, I would look deeper into how the [Python import system works](https://docs.python.org/3.7/reference/import.html), and make sure you {understand crontab](https://linux.die.net/man/5/crontab), or whichever tool you're using to accomplish the launch on startup. Also there are a few other [issues](https://github.com/Herring-UGAECSE-2920/Class-Instructions/issues?q=is%3Aissue+is%3Aclosed) and [FAQ questions](https://github.com/Herring-UGAECSE-2920/Class-Instructions/blob/main/FAQ.md) related to this you could take a look at.

ktkoffroth, 2021-04-09:
@jss83507 Did you guys have any luck resolving this issue?

jss83507, 2021-04-16:
We had some luck resolving the issue, our code will now run through the terminal without giving errors for package imports but we have not yet brought it all together to try running the scripts from startup.

closed 2021-04-16 by jss83507