Herring-UGAECSE-2920/Class-Instructions #30: Arc Command Validation

aarcowedu, 2021-04-16:
The gcode req's specify that we should validate arcs before attempting to draw them (referring to G02 and G03).
What should our program do when an invalid arc command is given? We've come up with a few possible solutions, but we're not sure which one to pick.
1. Stop drawing and display a bad gcode error
2. draw a straight line instead of an arc
3. skip to the next command and draw from its current position, leading the next line to also be wrong
4. use the arc endpoint as part of the next command, leading the next line to be right

We are heavily leaning towards solution 1 as it would make the most sense, but we can't find a specification in the req's.

ktkoffroth, 2021-04-16:
Hi @aarcowedu  there should not be any invalid arc commands in those files, I'll look into it, thanks

ktkoffroth, 2021-04-16:
@aarcowedu It appeared to be an issue with the arc offsets, changed it from 10 to 5. You can re-download the files and we'll make an announcement in class today.

closed 2021-04-17 by aarcowedu